workcraft/workcraft #1255: Copy user-defined assertions on model conversion

danilovesky, 2021-07-01:
On model conversion (when a new model is created from a given model), copy all the user-defined properties into the new model. For example, if an STG has a custom assertion, then copy this assertion into the synthesised circuit model.

vkhomenko, 2021-07-01:
This is tricky, e.g. the STG assertion can refer to STG places, which are not present in the circuit. Might be possible for some other kinds of properties which are guaranteed to refer only to signals.

danilovesky, 2021-07-01:
Yes, the assertion may become incorrect/irrelevant after copying into a derived model (especially if it refers to STG-specific elements). I do not plan to analyse these, but just blindly copy the whole assertion. It would be the user responsibility to modify/remove those assertions that become irrelevant. Maybe add a dialog or a global setting that determines if assertions should be copied?

vkhomenko, 2021-07-01:
This approach may lead to spurious failures which may upset new users - they may think the synthesis tools produced a wrong circuit as the assertions are no longer satisfied. Still worse, there could be subtle problems which don't lead to syntax errors but change the semantics of assertions, e.g. REACH allows one to use a regex to match e.g. place names, and it will "work" on the circuit-STG, but match something random.

vkhomenko, 2021-07-01:
On a more philosophical note, if we consider a hierarchical design, the assertions are attached to nodes in a hierarchy, and should not automatically propagate to nodes which are lower in the hierarchy, as the lower nodes may not have sufficient information to evaluate that assertion. E.g. if a node is a composition (i.e. block-level circuit), it may refer to elements of all its components, and so cannot be evaluated at any lower node in isolation. If the node is an STG and the lower node is its refinement (e.g. synthesised circuit) then information like STG places etc. disappears.

danilovesky, 2021-07-01:
I agree, this does not fit the philosophy of hierarchical design with refinements. Let us postpone this feature...

danilovesky, 2021-07-06:
I agree, this does not fit the philosophy of hierarchical design with refinements.

closed 2021-07-06 by danilovesky