workcraft/workcraft #1259: Squashing of circuit components

danilovesky, 2021-07-19:
Provide an API for _squashing_ a circuit component `c` that has another circuit `C.work` as a refinement model.
The content of `C.work` should be put into a page with a reference `c` (the same reference as the original component), and the connections to/from the pins of the component and to/from the ports of the  inserted circuit should be merged.

vkhomenko, 2021-07-19:
Is "squashing" a standard term? Like squashing commits in git? I'd consider "inline" or other terms, squashing for me implies destroying. At least it should be squashing hierarchy rather than a component.

danilovesky, 2021-07-19:
Not sure about the standard terminology, I just chose one that is good enough (squashing commits gave me an idea). It looks ok in the code too (I would avoid _inline_ because it reminds me C keyword).

vkhomenko, 2021-07-19:
Well, the meaning of "inline" in C++ is pretty much what you try to do - replace a call to a function (or module in case of a circuit/Verilog) by its body :-)

I guess "squash" can be used if you apply it to a subgraph, but if you apply it to a component than it's inlining.

danilovesky, 2021-07-19:
Inline sounds nice and clear for a software developer (better than squash), but may be confusing for the unprepared user.
Also note that it is not a full inlining, as the content of the foreign circuit will be kept in its page/namespace.

danilovesky, 2021-07-19:
@vkhomenko It looks like this operation is called _ungrouping_ in Synopsys tools. Too bad we already have group/ungroup operations in Workcraft.

PR #1260 (2021-07-21): Replacing circuit component by its implementation
Closes #1259
commit b061f794 (2021-07-21): Replace circuit component by implementation

Closes #1259
closed 2021-07-21 by danilovesky