workcraft/workcraft #1305: Unexpected name clashing with implicit STG places

danilovesky, 2021-12-09:
**To reproduce:**
1. Create the following STG: 
[image]
2. Convert `p1` into implicit place (via popup menu)
3. Rename `p0` into `p1`

**Observed behaviour:**
1. An unexpected warning appears (there should be no `p0` now): 
[image]
2. If the user clicks `OK` the error is produced:
[image]

**Expected behaviour:**
The place `p1` should bet renamed into `p0` with no warnings (as there should be no name clash with the implicit place).

vkhomenko, 2021-12-10:
There may be an issue with two implicit places having the same name, e.g. in the above STG, if you create a place p2 with arcs in+ ->p2 -> out+ and then hide both p1 and p2, they might get the same implicit name <in+,out+>

danilovesky, 2021-12-10:
Somehow Workcraft does not allow the second place to become implicit (the menu item is greyed out). Looks like we thought about this scenario before.

closed 2021-12-10 by danilovesky

vkhomenko, 2021-12-10:
Going via the main menu (not the popup menu) makes one of the places implicit while leaving the other explicit. It's probably surprising for the user and it would be better to allow parallel arcs like that, but I guess we can leave it as is - it's very low priority.

danilovesky, 2021-12-10:
Agree, it is not ideal, but very low priority.
And I have no idea how to "distinguish" such implicit places - they only can be referenced by the names of pred/succ transitions. Maybe they should merge into a single place (as they are redundant replicas of each other)?

vkhomenko, 2021-12-10:
MPSAT preserves the place with the smallest number of tokens and deletes the rest. Agree to leave as it is.

PR #1306 (2021-12-10): Prevent name clashing with implicit STG places
Fixes #1305
commit 0bb18d57 (2021-12-10): Prevent name clashing with implicit STG places

Fixes #1305