workcraft/workcraft #1311: Improve compatibility diagnostics between circuit component and its refinement model

danilovesky, 2022-01-05:
When assigning a refinement model to a circuit component the following compatibility issues should be explicitly communicated to the user:

* Component label is different from the refinement model title
* Component has pins that do not correspond to refinement signals
* Component misses pins that are present in the refinement model
* Input/output type of the component pins are different from the corresponding signals in the refinement model

vkhomenko, 2022-01-06:
Should some of these be warnings and other errors? E.g.:
  * "Component label is different from the refinement model title" - likely a warning, e.g. if you have a C-element block, its refinements could be named like C-majority-gate, C-Mayevsky, etc.
  * "Component has pins that do not correspond to refinement signals" - I think this is too broad, needs to be finer grained. I think the outputs must be the same (and it's an error if they are not), but for inputs we can allow subset, e.g. `(*) -> a+ -> b+ -> x+` can be refined by `(*) -> b+ -> x+` as input `a+` is unnecessary for computing output `x` (so this could be a warning rather than an error), or even silently accepted.

danilovesky, 2022-01-06:
Difficult to say what flexibility to allow, I am still experimenting.
* The main reason for refinement title matching component label -- these are used in Verilog export and necessary to match instances with their modules.
* Component pins may remain disconnected, but I would still prefer to have them though (collecting feedback from users about this).

vkhomenko, 2022-01-06:
re missing input pins, I think they must be allowed (maybe with a warning) - e.g. MPSAT may remove some dead signals during CSC resolution, synthesis, etc.

danilovesky, 2022-01-07:
A significant improvement is committed in #1313. Another iteration may be attempted based on user feedback at a later stage.

closed 2022-01-07 by danilovesky