workcraft/workcraft #1321: Option to suppress warning on loading work file created by newer version of Workcraft

danilovesky, 2022-01-31:
When a work file is open that created by a newer version of Workcraft the following warning is issued:
```
Workcraft vA.B.C may incorrectly read a file produced by newer Workcraft vX.Y.Z
Proceed with loading of work file '/path/to/file.work' anyway?
```

This message appears on every loading operation and may become annoying.
Most of the time it is safe to open such files.

A global preference for suppressing such a warning should be implemented.
Some flexibility in choosing when to suppress would be beneficial.
Alternatively some metadata should be saved in .work file specifying the oldest version that file is compatible with without the need for any warnings.

vkhomenko, 2022-01-31:
Maybe the file format should have its own version number?

danilovesky, 2022-02-01:
Not a bad idea.. I am just worried it will trigger a significant rewrite of compatibility manager that is "tight" to Workcraft version.
Ideally each model plugin should keep track of this file format version and "register" compatibility-related items based on that (versioning will vary for different models). Let me think about it.

danilovesky, 2022-02-07:
Enable model plugins to declare the oldest Workcraft version their deserialiser of work file is compatible with (in Major.Minor.Revision format). By default,  assume backward compatibility of the file format for any "Revision" as soon as "Major.Minor" part matches.

PR #1325 (2022-02-08): Save explicit compatibility version in meta data
Closes #1321
commit be67df09 (2022-02-08): Save explicit compatibility version in meta data

Closes #1321
closed 2022-02-08 by danilovesky