workcraft/workcraft #1334: Read-arcs not working in synthesized circuit

Definitely-Legit, 2022-03-10:
### **Please** answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?

Version 3.3.6

2. What operating system are you using?

CentOS

4. What did you do? If possible, provide a list of steps to reproduce
   the error.

I defined the following STG and synthesized an implementation with all 8 synthesis options:
[image]

5. What did you expect to see?

I expected the output to depend on inputs cs and wr in some way.

6. What did you see instead?

All 8 synthesis options generated a circuit whose cs and wr inputs are hanging. This is the result obtained with the "Complex gate [MPSat]" syntesis option:
[image]

danilovesky, 2022-03-10:
The synthesis result is correct - `out` only depends on `dis_0` according to the STG. As `dis_0` is an input, those read arcs that are supposed to prevent `dis_0` switching are responsibility of the environment and the circuit cannot "implement" them.

closed 2022-03-10 by danilovesky

Definitely-Legit, 2022-03-10:
My bad, thank you for the explanation.