workcraft/workcraft #1339: Check for missing phases of input signals before conformation check

danilovesky, 2022-03-18:
N-way conformation (and 1-way conformation) relies on parallel composition of several STGs. If one of the STGs misses a signal phase that signal phase disappears from the composition STG. This may lead to unexpected verification results in the following case:

* STG `A` has output `x` with both phases present and STG `B` has input `x` with a single phase, say `x+`
* Composition of `A` and `B` will miss `x-` and its _shadow_ copy, which leads to incorrect conformation check:
STG `A` may reach a state where `x-` is issued, that is not expected by STG `B` (clear violation of conformation), however, this would not be detectable in the composition STG (as it does not have `x-`)

A solution is to structurally detect the above situation and output an error (with detailed information about missing phases of input signals) for conformation checks. The precondition for the error is as follows: one STG has output `x` with both phases and another STG has `x` as an input with a single phase.

The error message should read as follows:
```
Model 'ABC' misses one of the phases for input signals:
  • 'x' has rising phase but not falling
  • 'y' has falling phase but not rising
```

vkhomenko, 2022-03-18:
I think if x is an input in both STGs, it's still an error. E.g. one STG can do x+ -> x- -> z+ (z is output) and the other can do x+ -> a+ -> z+ (z+ is an input here), the violation of N-way conformation will be hidden due to missing x- in the composition.

danilovesky, 2022-03-18:
@vkhomenko Makes sense..

Do you have a counter-example for the case when one STG has `x` as an input with both phases, while the other STG has `x` as an output with one phase? I think this is ok to do conformation check on such STGs, but would appreciate your _input_ :)

vkhomenko, 2022-03-18:
x+ -> x- -> z+ (x is input, z is output) and x+ -> a+ -> z+ (x is output, a,z are inputs)

danilovesky, 2022-03-18:
@vkhomenko I have changed the error message as follows (probably can be worded better):
```
Model 'ABC' misses one of interface signal phases that are present in another model:
  • 'x' has rising phase but not falling
  • 'y' has falling phase but not rising
```

vkhomenko, 2022-03-18:
Rising/falling phases may be too cryptic, what about:
A has x+ but not x-
B has x+ and x-
C has x- but not x+

danilovesky, 2022-03-18:
That would be good without signal renaming. This signal `x` in A may be called `y` in B and `z` in C.

vkhomenko, 2022-03-18:
How about:
Model 'ABC' misses phases of interface signals that are present in other models:
  • there is x+ but no x-
  • there is y+ but no y-

commit df886a2f (2022-03-18): Check for compatibility of interface signals

Before verification of conformation property do the structural checks
for missing phases and mismatch initial state of interface signals.
This is to prevent removal of signal events in composition STG which
may lead to incorrect verification result.

Closes #1339
closed 2022-03-18 by danilovesky