workcraft/workcraft #1350: SR latch initialisation not working properly in Digital Circuit plugin

Definitely-Legit, 2022-05-03:
Please answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?

Workcraft 3.3.7

2. What operating system are you using?

CentOS release 7.9.2009

3. What did you do? If possible, provide a list of steps to reproduce
   the error.

I defined a gate library in the SIS Genlib format which includes a SR latch, as shown in the image below. I'm using it for technology mapping with MPSat.

4. What did you expect to see?

The SR latch is defined as follows:
[image]
When both inputs are high the output is supposed to be high as well.

5. What did you see instead?

When using the Initialisation Analyser with a circuit that uses these latches and both inputs are initialised to high, the output is not shown as propagated high as I was expecting.
[image]

danilovesky, 2022-05-03:
This is a bug in the interpretation of GenLib sequential functions (thank you for reporting!).
It happened because of a confusion between set/reset functions in Workcraft and set/rest pins of a latch.

For the above function `q = s + !r * Q_NEXT` Workcraft incorrectly extracts `Set=s` and `Reset=r`. (This is based on the assumption that sequential functions are in the form `Q = Set + !Reset * Q_NEXT` which is used by Petrify and MPSat to output sequential synthesis results).

Instead Workcraft should do the following:
`Set = s + !r * [Q_NEXT=0] = s`
`Reset = !(s + !r * [Q_NEXT=1]) = !(s + !r) = !s * r`

As a temporary solution I would suggest to declare your latch in GenLib file as follows:
`q= s + !(!s * r) * Q_NEXT`

This should render the latch and its initialisation on all-high inputs as follows:
[image]

Definitely-Legit, 2022-05-04:
Thank you for the temporary solution, I can confirm that it is working.

commit f59c9e2d (2022-05-10): Fix parsing of set/reset in GenLib latch function

Fixes #1350
closed 2022-05-10 by danilovesky