workcraft/workcraft #1356: Exclude some sequential gates from conversion into resettable counterparts

danilovesky, 2022-05-19:
There may be sequential gates in the library without resettable counterparts. These gates should be initialised via their inputs (may not be always possible).

**Option A (arbitration primitives):** Currently the initialisation problem is with arbitration primitives (MUTEX and WAIT). It may  be a good-enough solution to forbid/allow resetting this specific type of gates.

**Option B (forbid/allow list):** Have a list of sequential gates that have resettable counterparts with default set/clear pins.

**Option C (external config):** Put information about initialisation of sequential gates in a separate config file, where details of gate renaming and its set/clear pins can be flexibly specified.

vkhomenko, 2022-05-19:
Some thoughts:
  * it is possible to work out whether there is a set or reset pin from the equation
  * if the latch is not mapped, we probably can handle it as if there is a variant with set/reset pin
  * for arbitration primitives there are some further restrictions, e.g. WAIT should not be initialised with `ctrl=1`, and mutex should not be initialised with 11 on outputs

danilovesky, 2022-05-31:
Decided to go with a mixture of **Option B** and **Option C**:
-  Add two global settings for Digital Circuit model, one for converting gates into init-low, the other for conversion into init-high  counterparts
- Each setting is in the form of-comma-separated list of *original_gate->replacement_gate(init_pin)*, e.g. *C2->C2_R(rst), NC2->NC2_R(rst), MUTEX->MUTEX_R(rst)*
- Only those mapped sequential gates present in these lists get converted into resettable counterpart (unmapped gates still get converted by modifying set/rest function)