workcraft/workcraft #1384: Support to transition inhibitors in Petri Nets

samjohn24, 2022-09-10:
Hi,

First, I would like to thank you for this tool, petri nets plugin is great for system simulation and analysis.

I see that in 3.3.8 version is not possible to convert a transition net to an inhibitor one. It is possible to add support for it?

Thanks!

danilovesky, 2022-09-11:
Extending Petri nets with inhibitor arcs should be doable for drawing and simulation purposes. However, verification of such models is non-trivial, and would require significant amount of work in MPSat backend. Also if we expand inhibitor arcs to STGs, then synthesis would affected. 
@vkhomenko What do you think?

vkhomenko, 2022-09-12:
For safe (or k-bounded) nets it is always possible to get rid of inhibitor arcs using complimentary places. For unbounded nets, it is not possible in general, since unbounded PNs with inhibitor arcs are Turing-powerful while ordinary PNs are not.

Since we require promised capacity, we deal with k-bounded nets, so it is theoretically possible to add inhibitor arcs. However, inhibitor arcs would be a change in the very core data structures, file formats, and utilities like checking the enabledness of a transition, finding preset or postset of a node (sometimes one needs to include inhibiting places, but not always), so adding inhibitor arcs would require touching the code in many places. There will be technical problems to solve, e.g. the user may want to refer to inhibiting places in properties, so if we remove inhibitor arcs using complementary places, the properties may need to be amended before they are fed to backends. 

So it is theoretically doable but it's a major effort. I think for now we can just recommend the users to use the complementary places construction manually.

danilovesky, 2022-09-12:
Let us log this as a low priority feature, and maybe tackle it at the same iteration as #943.