workcraft/workcraft #1409: Optimise the usage of line breaks at some places

elfring, 2022-12-09:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
I noticed that line breaks were appended to strings and they were stored within strings which contained only two characters for this purpose.

:thought_balloon: Thus I propose to [adjust affected string specifications](https://docs.oracle.com/en/java/javase/19/docs/api/java.base/java/lang/StringBuilder.html#append(char) "Appending a single character") accordingly.

Would you like to integrate anything from a transformation result which can be generated by a command like the following?
```
lokal$ perl -p -i.orig -0777 -e $'s#\\+\\s*\"\\\\n\"#+ \'\\\\n\'#gm' $(find ~/Aktivitäten/Workcraft/lokal -name '*.java')
```

:crystal_ball: How will the development interests evolve further also according to update candidates in 88 lines of this software?

danilovesky, 2022-12-09:
No objections to this tidying up of the code.

closed 2023-04-27 by danilovesky

elfring, 2023-04-27:
Thanks for [your integration of further source code improvements](https://github.com/workcraft/workcraft/commit/afa25e6a8d2f70f8fcb0eca55bd311d65a01a66e "Refactoring for string concatenations").