workcraft/workcraft #1411: Support Apple M1 chip

pkriens, 2023-01-15:
1. What version of Workcraft are you using? Latest download Jan 2023

2. What operating system are you using? MacOS 12.6 on M1 chip

3. What did you do? If possible, provide a list of steps to reproduce
   the error. Built in commands fail

Is there any intention to support the Apple M1 (aarch64) version?

vkhomenko, 2023-01-16:
Hi, Workcraft depends on a number of 32-bit backend tools which would be difficult to re-design for 64-bit architectures. As OSX no longer supports 32-bit executables, it cannot run Workcraft. However, you can still run it from a virtual box, see [https://workcraft.org/help/virtualbox](url).

closed 2023-01-16 by vkhomenko

pkriens, 2023-01-16:
I was afraid of this but it locks out a major part of the potential market. Since SAT has a Java version, how much work would it be to port the tools to Java?

vkhomenko, 2023-01-16:
There are tons of C/C++ code there, with many dependencies. I don't think it's realistic to rewrite everything in Java. It is more realistic to port backends to 64-bit, but even that would be a huge effort.

danilovesky, 2023-01-16:
@pkriens Unfortunately the legacy C code backend tools rely on is rather difficult to port, as it uses 32-bit pointer arithmetic. Just recompiling it for non-i386 architecture causes "intermittent" issues which are a pain to debug. I must admit we never tried asm.js (and do not have bandwidth to attempt it now), but I expect many issues on that pathway. Even if it works out smoothly, the performance degradation would kill all the fun.

Currently the plan is to drop OSX support (as 32-bit compatibility requirement makes this effort useless), and advise OSX users to run Workcraft in a virtual machine, e.g. https://workcraft.org/help/virtualbox (I hope 32-bit code still runs well in guest OS).