workcraft/workcraft #1440: Support scan ports on mutex cells

omer-qadir, 2023-03-06:
Using workcraft version 3.3.9 on Windows and Linux. This is not a bug, but a feature request. The mutex cell can be configured using the preferences, but this does not allow using a mutex std-cell with additional scan ports. This does not seem possible using the substitution library either, which means it requires a manual edit of the exported verilog to add scan ports.

danilovesky, 2023-03-07:
Could you elaborate a bit about the potential use of this feature?

Do you have a specific implementation of testable MUTEX in mind? Is it based on ASYN paper form 2015?

Also, do you think this can be achieved by extending substitution rules?

For example, if we support a substitution rule like this, allowing creation of pins that were not in the original cell:
`MUTEX=MUTEXD2 (r1=R1, r2=R2, g1=G2, g2=G2, =SI, =SO, =SE)`

then the export of a standard Workcraft mutex `me` connected to `r1`/`r2` requests and `g1`/`g2` grants would look like this in a Verilog file:
`MUTEXD2 me (.R1(r1), .G1(g1), .R2(r2), .G2(g2), .SI(), .SE(), .SO());`

Is it something useful (with scan pins being unconnected)? Or do you envision this mutex to be stitched into a scan chain too?

omer-qadir, 2023-03-07:
I have a MUTEX standard cell implementation, which is not based on Yang Zhang paper in async 2015. Hopefully can publish something soon, but for now it is not in the public domain yet. It is stitched into the scan chains and today we need to manually edit the files and do the stitching. 

Ideally it would fit into the Workcraft flow for scan-insertion, but if that is too difficult to implement, then certainly the substitution rule extension would be a good improvement as well. It should however, support connecting an arbitrary signal to a port as well, so that no manual editing is required. For example the following type of rule could work:

`MUTEX=MUTEXD2 (r1=R1, r2=R2, g1=G2, g2=G2, "scanIn"=SI, "scanOut"=SO, "scanEn"=SE)`

where the quotes could be the indication that this string is not an existing port in the MUTEX cell and should be copied verbatim...?

danilovesky, 2023-03-07:
Let us try to exhaust substitution rules pathway first (as it would be much easier to implement).

What if we have several mutex elements, say `me1` and `me2` connected to some requests and grants?

If the substitution rule is as follows:
`MUTEX=MUTEXD2 (r1=R1, r2=R2, g1=G2, g2=G2, "scanIn"=SI, "scanOut"=SO, "scanEn"=SE)`

then the produced Verilog would connect both `me1` and `me2` to the same set of scan ports:
```
MUTEXD2 me1 (.R1(...), .G1(...), .R2(...), .G2(...), .SI(scanIn), .SE(scanEn), .SO(scanOut));
MUTEXD2 me2 (.R1(...), .G1(...), .R2(...), .G2(...), .SI(scanIn), .SE(scanEn), .SO(scanOut));
```

Would this be acceptable?

omer-qadir, 2023-03-07:
Yes that would be acceptable. The scan stitching is done by a backend tool. But you make a good point and if it's equally easy to implement, it would be great to support both ways of hooking up new signals. Namely, leaving them unconnected or hooking them up to a named signal.

danilovesky, 2023-05-10:
@omer-qadir I am thinking of implementing this as an optional _string_ at the end of the substitution rule.
Your above example would be expressed as follows:
`MUTEX=MUTEXD2 (r1=R1, r2=R2, g1=G2, g2=G2) ", .SI(), .SO(), .SE(scanEn), .RN(1'b1)"`

Whatever string is found between the double-quotes (including spaces, commas and other symbols) will go after the last connection of the cell. Does it look reasonable?

closed 2023-05-10 by danilovesky

omer-qadir, 2023-05-15:
Apologies for the slow response. Yes this looks great thanky you for the fix!