workcraft/workcraft #1518: Promote absence of local self-triggering in STG to essential properties with exceptions

danilovesky, 2023-08-22:
Absence of local self-triggering defined in #1516 should usually hold, with rare exceptions, such as pulse generator STGs. Therefore this property should be promoted to essential property with a GUI to define exceptions.

Reach expression with example exceptions:
```
card DUMMY != 0 ? fail "Absence of self-triggering can be checked only on STGs without dummies" :
let
    EXCEPTION_NAMES = {"out1", "sig2", ""} \ {""},
    EXCEPTIONS = S EXCEPTION_NAMES
{
    exists s in LOCAL \ EXCEPTIONS {
    let s_tran=T s {
        exists t1 in s_tran, t2 in s_tran s.t.
            t1!=t2 
            &
            ~ is_empty (pre t2 * (post t1 \ pre t1))   // t1 structurally triggers t2
            &
            is_empty (pre t2 * (pre t1 \ post t1))   // t2 is not disabled by t1 
        {
            @t1 & ~@t2
            &
            forall p in pre t2 s.t. ~(p in post t1) {
                ~(p in pre t1) & $p
            }
        }
    }
} 
```

vkhomenko, 2023-08-22:
"Absence of [local] self-triggering ..."?

danilovesky, 2023-08-22:
I guess any self-triggering check would rely on no-dummy assumption. Anyway, I will add "local" for consistency.

danilovesky, 2023-08-29:
Improvement: reduce the number of local signals in the GUI to syntactic self-triggers only.

commit a17c6ec5 (2023-08-22): Improve local self-trigger check with exceptions

Closes #1518
commit 0f952f65 (2023-09-01): Reduce number of signals to syntactic self-trigger

Close #1518
closed 2023-09-01 by danilovesky